linux-surface/intel-precise-touch #1: The new ipts driver implementation breaks gnome shell

alexanderi96, 2020-01-25:
Hi there, Fedora 31/Surface Pro 5 user here.
After installing the new driver and rebooting my device i noticed that gnome shell stops responding correctly to my inputs after a couple of touches and swipes on the icons.
After the bug is engaged gnome shell stops responding to the mouse inputs and the gnome shell bar only responds if the win button is pressed taking me to the opened app overview page as usual.
If I start clicking repeatedly on the bar (only through the mouse) it acts like an "Alt+Tab" trigger and switches to the next opened app without any animation.
I don't have to report any problem for the other app. Firefox keeps reciving the touches correctly and xournalpp seems faster as well while writing with the pen.

StollD, 2020-01-25:
This sounds more like a Gnome bug to me than an IPTS bug to be honest. All applications receive the same touch events from the kernel, and if firefox etc. interpret it correctly, it seems that gnome-shell doesn't.

I remember having the same problem before with the old IPTS driver as well.

alexanderi96, 2020-01-27:
It makes sense, strange that i didn't hat this issue before.
I'm closing this issue then!

closed 2020-01-27 by alexanderi96