linux-surface/intel-precise-touch #2: Pressure seems not to be working

olantrust, 2020-04-16:
Using Surface Kernel 5.6.4 (Manjaro with KDE on Surfacebook 2)
Pen is working, but no pressure sensitivity appears to work.

olantrust, 2020-05-01:
@StollD @qzed is pressure meant to be working with this implementation, or is it a desired feature?
Apologies, but it wasn't stated anywhere.

StollD, 2020-05-01:
Pressure does work for me in Xournal++ (using Gnome + Wayland). You can check if the pressure events are sent correctly by running `sudo evtest | grep ABS_PRESSURE` and then selecting the IPTS Stylus device.

olantrust, 2020-06-29:
Hi indeed that's working, with Microsoft stylus, but not the others.
However it has no impact using Krita or KolourPaint. Will check into KDE issues :s

StollD, 2020-06-29:
Weird. I have pressure working just fine in Krita, under Gnome even (both Wayland and X11). I'd imagine that if it works under Gnome, it will work under KDE too.

Maybe you need to configure something in Krita? I saw that it has some tablet related settings, including a tablet tester. Maybe that has some knobs you need to turn.

Regarding third party styli: Do you know if these have working pressure under windows? If yes there might be some protocol difference that we need to figure out. But it wouldn't suprise me if the hardware simply doesnt support it.

closed 2021-02-15 by StollD