linux-surface/intel-precise-touch #3: Module does not work when intel_iommu=on is set

libanp, 2020-05-09:
When `intel_iommu=on` is set not touch or stylus events are received even with `iommu=pt` also set. `evtest` does not show any events and gnome shell does not respond either. `modprobe -r ipts` does not remove the device file from `/dev/input/` and programs that try to access the old device file hang.

Expected behaviour: Working singletouch or stylus input; Clean module removal to allow switching between touch modes.

Kernel: 5.6.11-surface (behaviour was the same when I tested 5.5.13 and 5.6.7)
OS: Debian Unstable
IPTS firmware: 20200402-1

tiagocpontesp, 2020-05-12:
I didn't go through testing, but can confirm it just happened on my SB2.

StollD, 2020-06-03:
I thought about this a bit, and my only idea why this would happen is because we are using a "virtual" bus (MEI). The MEI driver creates an interface and a virtual device that is not backed by an actual device. Everything that is sent over this bus is wrapped with other data and sent to the actual ME device over the PCI bus.

My theory is that because we are using a virtual device, the IOMMU doesn't know which device the memory that we need to allocate for communicating with the ME should be mapped to, and the allocation fails. When the IOMMU is disabled it doesn't matter because we simply get a random chunk of memory + a physical address, with no device mapping being required.

I don't really know much about the IOMMU, but if that theory is true, I don't really think we can do a lot to fix it, simply because we don't have any reference to the actual physical device.

closed 2022-04-25 by qzed