linux-surface/intel-precise-touch #11: Request gen 7 pen support drivers from Intel?

obswwf, 2021-06-05:
Intel has a reputation of being quite open-source friendly, so would it be possible to write them a letter requesting the surface pen tracking drivers for gen 7 devices?

StollD, 2021-06-05:
Intel has never cared about IPTS. They only care about linux support for their hardware if they can sell it to someone. And they cant sell IPTS on Linux to someone, so they dont care.

But more importantly, they cant release the code for processing. The IPTS hardware is designed to make processing the job of the OEM, in our case Microsoft. Intel can release the documentation for how to interact with the hardware, but if you want to do anything with the data you receive, you need to get the algorithms from Microsoft.

obswwf, 2021-06-05:
> Intel has never cared about IPTS. They only care about linux support for their hardware if they can sell it to someone. And they cant sell IPTS on Linux to someone, so they dont care.
>
> But more [...]

Well that's sad to know…I recently installed arch on my SP7. Is there anything I can do to help with reverse-engineering pen support?

qzed, 2021-06-05:
If you really want to dig in: IIRC the `ipts-dbg` tool should be able to dump raw data. https://github.com/linux-surface/intel-precise-touch/tree/feature/fwd-to-userspc/logs/datadumps has some annotated dumps that might help you get started working out the data format. I don't think we have any annotated dumps for the Gen7 devices yet though. So all on that is probably in https://github.com/linux-surface/intel-precise-touch/issues/4#issuecomment-786707435 and the following comments.

obswwf, 2021-06-06:
> If you really want to dig in: IIRC the `ipts-dbg` tool should be able to dump raw data. https://github.com/linux-surface/intel-precise-touch/tree/feature/fwd-to-userspc/logs/datadumps has some annotated dumps that might help you get started working out the data format. I don't think we have any annotated dumps for the Gen7 devices yet though. So all on that is probably in [#4 (comment)](https://github.com/linux-surface/intel-precise-touch/issues/4#issuecomment-786707435) and the following comments. Specifically [#4 (comment)](https://github.com/linux-surface/intel-precise-touch/issues/4#issuecomment-788963731) for IDs.

I'll try that out when my schedule opens up.

closed 2021-06-06 by obswwf