linux-surface/intel-precise-touch #13: A question about future development for touch/stylus aspects of the surface line.

thekswenson, 2021-07-16:
I have a SP4 and have been benefiting from the community work on the project since before the jakeday days. Enjoying using the [stylus labs write](http://www.styluslabs.com/) app for all my notes. The machine is on its last legs with the battery bulging the screen out when hot. I would like to make an informed decision about my next surface pro-like purchase. Here are some of the limited options I can think of:

- SP6 (well supported, but hard to find now and rather expensive when available)
- SP7 (single touch support)
- SP7+ (best battery life but no touch or stylus support)
- SP8 (unclear, maybe can't wait)
- Dell latitude 7320 detachable (not clear how well linux runs on this) 

Could you folks please give me your best guess on potential future support for each of these machines?
This is what I know after poking around here:

- SP 7: it's a work in progress (https://github.com/linux-surface/intel-precise-touch/issues/4) but I'm not sure how hopeful people are for the stylus
- SP7+: it looks far off (https://github.com/linux-surface/intel-precise-touch/issues/8). is it safe to assume that nobody here plans on working on this?

Any comments or advice that you have on these matters (even on the DELL machine) would be greatly appreciated! Thank you all for you great work.

qzed, 2021-07-16:
Yeah that sounds about accurate. So on the SP7 I think you should already be able to get multi-touch (or enabling multi-touch shouldn't be that difficult as we now know how to get the data) but pen/stylus is still an issue there. That data seems to be provided in some sort of raw format that we don't know how to process yet. So someone with a device that emits that format needs to figure that out.

Note: I have a Surface Pro X that does seem to do that (at least based on a very quick look) but that device as a _ton_ of other (more important) problems (e.g. nvme SSD, ...). I haven't even gotten that to provide that data on Linux and had to resort to Windows to check that part, so my hope is someone else will have a look at this sooner than I'll probably be able to (also SPX stuff is currently a bit on hold since I'm dealing with other things).

The SP7+ uses a new Intel PCI device for sending the touch data (previously that was based on the Intel Management Engine), so for that we need an entirely new PCI device driver. That will either require a lot of reverse engineering or Intel stepping up and providing that driver (or at least a code dump that we can work with).

The SP8 will very likely be similar to the SP7+, unless MS decide to go with a SPI-over-HID interface which they use on the SPX and the AMD Surface devices (a driver for that interface exists, but needs to be ported to ACPI). Even then, it's very likely that the stylus data is still in the "new" format, i.e. needs some processing that we need to figure out. So for those devices there are essentially two problems that need to be solved.

The Dell you've mentioned probably (like almost every other non-Surface device out there, there are 2 exceptions IIRC) doesn't use IPTS (or heatmap/raw touch data), so it'll probably have better Linux support regarding touch/stylus. I can only speculate here though, as I don't actually know that device.

Can't really give you any time estimates. Right now there's little movement since none of us core devs have devices to properly investigate this. Hope I could give you an overview of what needs to be done though.

thekswenson, 2021-07-20:
Thanks for the summary @qzed !

I am happy to hear multitouch on the SP7 is nearly working. Given support for SP7+ and SP8 is probably far off (barring the miraculous appearance of help from intel -- I don't remember any help coming from them since the original very basic driver released by one of their devs so many years ago), I'm probably going to get a SP7.  

The DELL machine needs work and I'm reticent to try to learn (it's a maze, not even sure how I would learn this) the gritty details about getting the thing working (https://www.reddit.com/r/linuxhardware/comments/oceu59/linux_on_dell_inspiron_7320_detachable/).:
- the display is not working well
- cat /sys/power/mem_sleep gives only "s2idle" (does this mean deep sleep is impossible?)

closed 2021-07-20 by thekswenson

thekswenson, 2021-07-20:
In case someone else stumbles upon this thread...  there also seems to be a Lenovo detachable that works quite well with linux and has very good battery life:

https://www.reddit.com/r/thinkpad/comments/mdqxhf/x12_detachable_linux/

qzed, 2021-07-20:
> * cat /sys/power/mem_sleep gives only "s2idle" (does this mean deep sleep is impossible?)

That's also the same on the SP7 (or all Surface Pros, really). They all only have "connected standby", so I wouldn't count that against the dell.