linux-surface/intel-precise-touch #16: Pen calibration

han-so1omon, 2022-01-31:
I'm using Surface Book 2 with Manjaro 21.2 GNOME

My pen is functioning as if only the top-left quadrant of the screen was the entire screen. It tracks movements there, and then it multiplies all of those movements by 2 to position the cursor. For example. In the top left it is perfect. In the center of the screen it is the bottom right. Halfway down the screen on the left side it is all the way down on the left. Halfway across the screen on the top it is all the way to the right on the top.

It was working fine until I went to the Settings -> Wacom Tablet -> Calibrate, then calibrated using the pen. Calibration did not work with just my finger. After calibration, I am experiencing the effects described above.

What might be causing this issue and how can I fix it?

gattolfo98, 2025-03-24:
Same problem here on SP8

gattolfo98, 2025-03-24:
I have a solution, i'm using gnome, i installed with aur libwacom-surface and now in the setting i can calibrate the pen, so after the calibration now the pen point in the correct point

antoniovazquezblanco, 2026-11-07:
As per https://github.com/linux-surface/kernel/pull/178#issuecomment-6447636724, it seems that the correct place to send PRs and bug reports is now https://github.com/linux-surface/kernel.

This repo should be archived or updated by the maintainers!