linux-surface/intel-precise-touch #18: Sensor keeps reseting after load ipts module

Christoxygen, 2022-07-25:
after `sudo rmmod ipts` and `sudo insmod ipts.ko`, ipts keeps reporting sensor was reset and restarting like below, where the problem might be? I'm using Fedora 36 with kernel from linux-surface latest release.
[image]

StollD, 2022-07-25:
Yeah, this is currently an issue on the HID branch for gen7 devices. I think I know how to fix it, thanks to someone debugging it with me on matrix, I just havent done the required changes yet.

StollD, 2022-07-28:
This issue should hopefully be fixed on the HID branch now.

Christoxygen, 2022-07-28:
The great thing is that now it can respond to my touch but the response is very slow like when dragging a window, it takes about 0.7s to catch the finger. And the response is like my finger is always on the screen even if I have actually put my finger aside. Sometimes there is error message says `failed to read buffer : -1` in dmesg

StollD, 2022-07-28:
Could you post the full dmesg? Also, do you have iptsd active or are you testing the singletouch mode?

Christoxygen, 2022-07-28:
[dmesg.log](https://github.com/linux-surface/intel-precise-touch/files/9212083/dmesg.log)
I have iptsd activated

StollD, 2022-07-28:
Stop iptsd to enable singletouch mode.

Also the reset command seems completely cursed on gen7 ...

Could you maybe try this change? Download the attached file into the directory where you run `make` and run `git apply reset.txt`. After that please build the module, load it, maybe touch the display once, and then unload it again and upload the dmesg.

[reset.txt](https://github.com/linux-surface/intel-precise-touch/files/9212486/reset.txt)

Christoxygen, 2022-07-28:
[dmesg1.log](https://github.com/linux-surface/intel-precise-touch/files/9212722/dmesg1.log)
Still the same. And there is no touch response without iptsd

Christoxygen, 2022-07-28:
It says `reset.txt:31: space before tab in indent.
	 	header->cmd_type = IPTS_FEEDBACK_CMD_TYPE_SOFT_RESET;
`when applying reset.txt

Christoxygen, 2022-07-28:
And there is no touch response without iptsd running

StollD, 2022-07-28:
Okay, I think I somewhat know whats going on now.

Could you do me a favor? I am trying to figure out a way how I can distinguish the gen7 devices from the older ones so I dont always have to find solutions that work across the board. I would be interested to know what some of the device info variables that IPTS returns are on gen7. Could you build the module from the `eds` branch of this repository, load it, and then upload the dmesg again?

Christoxygen, 2022-07-28:
[Eds.log](https://github.com/linux-surface/intel-precise-touch/files/9213352/Eds.log)

StollD, 2022-07-28:
Thank you!

I pushed some more changes to the HID branch, could you try them and see if anything improves? (with and without iptsd)

Christoxygen, 2022-07-28:
Multitouch is unstable, sometims it just doesn't work but when it does, it works just fine (gnome gets stuck sometimes but I think it's gnome's problem)
Singletouch is still the same `failed to process buffer: -1`. 
And there aren't more messages in dmesg after turning iptsd on

Christoxygen, 2022-07-28:
btw how can i make it response to the touch of larger size

StollD, 2022-07-29:
> Multitouch is unstable, sometims it just doesn't work but when it does, it works just fine (gnome gets stuck sometimes but I think it's gnome's problem) Singletouch is still the same `failed to process buffer: -1`. And there aren't more messages in dmesg after turning iptsd on

Thank you again! Yes, GNOME can sometimes get a bit crazy with the touchscreen. Make sure that you dont have two instances of iptsd running by accident, I managed to do that in the past and got a completely unusable touchscreen.

Regarding singletouch, it seems that switching the module to using the native HID descriptor of the device (instead of a hardcoded fallback descriptor) might have caused this issue, because the native descriptor for singletouch is broken. I pushed another change that will hopefully fix the descriptor and make singletouch work.

> btw how can i make it response to the touch of larger size

Sadly this is currently hardcoded in iptsd: https://github.com/linux-surface/iptsd/blob/master/src/contacts/basic/processor.cpp#L26-L33

You could try switching to the advanced touch processing to see if that responds better to your fingers. Put this into `/etc/ipts.conf`:

```
[Touch]
Processing = advanced
```

Christoxygen, 2022-07-29:
still the same glitch

StollD, 2022-07-29:
Alright sorry, I made a mistake. I thought the descriptor that the hardware gives you would already contain the singletouch descriptor, but after checking the descriptor someone else dumped a while ago, thats actually not the case. So me fixing the descriptor obviously wont work.

I pushed a new fix that should hopefully make singletouch work now, and get rid of the error message. Thank you for being patient with me so far and helping debug this.

Christoxygen, 2022-07-29:
No big deal. However, like multitouch the singletouch is unstable, sometimes it needs several times of reloading ipts to make it work. But after all, it can work now. Thanks so much for the hard work.

antoniovazquezblanco, 2026-11-07:
As per https://github.com/linux-surface/kernel/pull/178#issuecomment-6447636724, it seems that the correct place to send PRs and bug reports is now https://github.com/linux-surface/kernel.

This repo should be archived or updated by the maintainers!