linux-surface/intel-precise-touch #19: Mmap and align heatmap to 4096-byte boundary for AVX512

danielzgtg, 2022-10-10:
This would improve the efficiency of [ipts#88](https://github.com/linux-surface/iptsd/issues/88).

It would be nice to align the heatmap portion of the messages to 4096-byte boundaries or whatever AVX512 needs. This will avoid either bus fault crashes, or slow unaligned accesses. I am currently relying on the workaround of doing a manual copy loop before passing things to AVX512. It would save battery and speed things up if I could avoid this copy.

To be precise, the heatmap specifically, not the whole buffer needs to be page-aligned. This means I imaging something like allocating two contiguous pages and telling the hardware to place the buffer a little before the middle of the two pages. This way I can mmap and do AVX512 directly without copying. If the hardware supports this, it would be nice if this driver provided this alignment.

qzed, 2022-10-10:
We're moving away from the whole custom device thing in favor of using hidraw. That better fits with how the newer devices work (they do directly provide HID data, the heatmap data is essentially just a custom field wrapped in that). Also some devices (Surface Pro X, the new ITHC stuff, likely also the AMD devices) directly provide a HID device. So this helps unifying all those different kernel drivers into one single already existing user-space API.

With how HID data is delivered to user-space you can't avoid copying that if you want it aligned. Especially as you need to parse the format and don't have any guarantees to find the heatmap at some fixed offset. I believe even with the old format you'd need to a copy for alignment since you also can't rely on it being at a specified offset there. Whether that's done in kernel or in user-space shouldn't matter.

danielzgtg, 2022-10-10:
> don't have any guarantees to find the heatmap at some fixed offset

Oh well.

closed 2022-10-10 by danielzgtg