linux-surface/intel-precise-touch #21: Thanks and Lots of Hidraw devices

jordanpwalsh, 2023-01-21:
First I wanted to thank @StollD for all the recent commits in January, some of the bug fixes have really increased the stability of the ipts driver, especially resuming out of suspend on a Surface Pro 7. 

One issue, however, is the driver seems to create a new hidrawXX device after resuming from suspend and these are piling up after some time. Is there anyway to resolve that?

Thanks again,
Jordan

StollD, 2023-01-21:
Could you please post your dmesg after suspending a few times, so that it creates some devices?

jordanpwalsh, 2023-01-22:
I attached two dmesg captures - one with type cover and the other without because the first one had so many listed I wanted to make sure that wasn't part of the problem. I believe these lingering devices are causing some kind of other issue as well because the system won't cleanly shut down due to being unable to kill the iptsd processes.

[dmesg_no_type_cover.txt](https://github.com/linux-surface/intel-precise-touch/files/10475460/dmesg_no_type_cover.txt)
[dmesg_with_type_cover.txt](https://github.com/linux-surface/intel-precise-touch/files/10475461/dmesg_with_type_cover.txt)

StollD, 2023-01-22:
When did you last update? The errors in the log should have been fixed by the earliest commits I made in January.

i.e. this:

```
ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Failed to request flush: -4
```

would trigger an infinite loop with this commit: https://github.com/linux-surface/intel-precise-touch/commit/404700ac27a2f105a5b42b9353e1ec63ae9a8b9b

and should not happen at all with this commit: https://github.com/linux-surface/intel-precise-touch/commit/862d095abd8714a635ad7711d62f6fe4ecc61405

jordanpwalsh, 2023-01-22:
> would trigger an infinite loop with this commit: 404700a
> and should not happen at all with this commit: 862d095

I have commit f4dd578 from 20-Jan checked out and built. I suppose it's
possible I miffed something with the install, but a little hard to tell
with the version being the same. I should have the latest of both ipts
driver and iptsd from the other repo, and some of my previous issues did go
away. I can bump the version to verify that's what I'm running unless you
have other suggestions.

StollD, 2023-01-23:
Maybe try to uninstall any versions of the module you have installed via. DKMS etc.

Then reinstall the kernel and finally the module.

Because I believe that running `sudo make dkms-install` won't actually update the module if it is already installed.

jordanpwalsh, 2023-01-23:
> Because I believe that running sudo make dkms-install won't actually update the module if it is already installed.

It won't, but it errors out. I ended up removing it from dkms via `dkms uninstall` as well as manually removing files and rebooting (The module wouldn't reinsert until rebooting). 

Let me do that again with a version bump to ensure I'm running the latest. Thanks.

antoniovazquezblanco, 2026-11-07:
As per https://github.com/linux-surface/kernel/pull/178#issuecomment-6447636724, it seems that the correct place to send PRs and bug reports is now https://github.com/linux-surface/kernel.

This repo should be archived or updated by the maintainers!