linux-surface/intel-precise-touch #23: stylus (only) not not working in ubuntu 23.04

ECO1AI, 2023-04-28:
hardware : surface pro 4
distro : kubuntu , ubuntu (both 23.04)
kernel: 6.2.12-surface (64-bit)

problem occurred after upgrading kubuntu and using stylus once.

tried reinstalling : iptsd , libwacom-surface (with all dependencies) but no change.
also, tried libwacom-surface-dev but no change.
reinstalled OS twice.  
the stylus is not showing in tablet settings in both gnome and kde plasma.
the program that tries to find tablets crashes. 
however, touch is working normally.

`dmesg` :
```
[ 5784.853415] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Stopping IPTS
[ 5785.088053] ACPI: EC: interrupt blocked
[ 5785.088832] intel_pch_thermal 0000:00:14.2: CPU-PCH is cool [31C]
[ 5785.123470] ACPI Warning: Time parameter 255 us > 100 us violating ACPI spec, please fix the firmware. (20221020/exsystem-141)
[ 9290.832234] ACPI: EC: interrupt unblocked
[ 9291.206359] i2c_hid_acpi i2c-MSHW0030:00: failed to set a report to device: -121
[ 9291.206706] i2c_hid_acpi i2c-MSHW0030:00: failed to set a report to device: -121
[ 9291.207043] i2c_hid_acpi i2c-MSHW0030:00: failed to set a report to device: -121
[ 9291.207374] i2c_hid_acpi i2c-MSHW0030:00: failed to set a report to device: -121
[ 9291.207706] i2c_hid_acpi i2c-MSHW0030:00: failed to set a report to device: -121
[ 9291.208042] i2c_hid_acpi i2c-MSHW0030:00: failed to set a report to device: -121
[ 9291.208373] i2c_hid_acpi i2c-MSHW0030:00: failed to set a report to device: -121
[ 9291.208710] i2c_hid_acpi i2c-MSHW0030:00: failed to set a report to device: -121
[ 9291.209039] i2c_hid_acpi i2c-MSHW0030:00: failed to set a report to device: -121
[ 9291.287535] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Starting IPTS
[ 9291.287686] OOM killer enabled.
[ 9291.287692] Restarting tasks ... 
[ 9291.290376] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: IPTS EDS Version: 1
[ 9291.295693] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: IPTS running in event mode
[ 9291.296900] input: IPTS 1B96:006A as /devices/pci0000:00/0000:00:16.4/0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04/0000:1B96:006A.0005/input/input17
[ 9291.300129] done.
[ 9291.300163] random: crng reseeded on system resumption
[ 9291.469707] PM: suspend exit
[ 9291.498748] hid-multitouch 0000:1B96:006A.0005: input,hidraw0: <UNKNOWN> HID v0.00 Device [IPTS 1B96:006A] on 
[ 9357.918635] hid-generic 0005:1345:1002.0006: unknown main item tag 0x0
[ 9357.919208] input: Bluetooth Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/bluetooth/hci0/hci0:1/0005:1345:1002.0006/input/input19
[ 9357.920261] input: Bluetooth Keyboard Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/bluetooth/hci0/hci0:1/0005:1345:1002.0006/input/input20
[ 9357.922955] hid-generic 0005:1345:1002.0006: input,hidraw1: BLUETOOTH HID v1.1b Keyboard [Bluetooth Keyboard] on 98:5f:d3:c5:ad:88
[ 9657.089364] GpuWatchdog[8003]: segfault at 0 ip 00007fa540f927e6 sp 00007fa537530430 error 6 in libcef.so[7fa53caef000+7770000] likely on CPU 3 (core 1, socket 0)
[ 9657.089376] Code: 89 de e8 0d ef 6e ff 80 7d cf 00 79 09 48 8b 7d b8 e8 4e 66 2c 03 41 8b 84 24 e0 00 00 00 89 45 b8 48 8d 7d b8 e8 1a d4 b5 fb <c7> 04 25 00 00 00 00 37 13 00 00 48 83 c4 38 5b 41 5c 41 5d 41 5e
[ 9792.194471] perf: interrupt took too long (3197 > 3128), lowering kernel.perf_event_max_sample_rate to 62500
```
`sudo journalctl -p 3 -xb` :
```
11:08:26 ECO-Origami kernel: i2c_hid_acpi i2c-MSHW0030:00: failed to set a report to device: -121
أبريل 28 11:08:26 ECO-Origami kernel: i2c_hid_acpi i2c-MSHW0030:00: failed to set a report to device: -121
أبريل 28 11:08:26 ECO-Origami kernel: i2c_hid_acpi i2c-MSHW0030:00: failed to set a report to device: -121
أبريل 28 11:08:26 ECO-Origami kernel: i2c_hid_acpi i2c-MSHW0030:00: failed to set a report to device: -121
أبريل 28 11:08:26 ECO-Origami kernel: i2c_hid_acpi i2c-MSHW0030:00: failed to set a report to device: -121
أبريل 28 11:08:26 ECO-Origami kernel: i2c_hid_acpi i2c-MSHW0030:00: failed to set a report to device: -121
أبريل 28 11:08:26 ECO-Origami kernel: i2c_hid_acpi i2c-MSHW0030:00: failed to set a report to device: -121
أبريل 28 11:08:26 ECO-Origami kernel: i2c_hid_acpi i2c-MSHW0030:00: failed to set a report to device: -121
أبريل 28 11:08:26 ECO-Origami kernel: i2c_hid_acpi i2c-MSHW0030:00: failed to set a report to device: -121
أبريل 28 11:09:33 ECO-Origami systemd-udevd[18065]: hidraw1: Process '/usr/bin/iptsd-check-device --quiet /dev/hidraw1' terminated by signal ABRT.
أبريل 28 11:09:33 ECO-Origami systemd-udevd[18065]: hidraw1: Failed to wait for spawned command '/usr/bin/iptsd-check-device --quiet /dev/hidraw1': Input/output error
أبريل 28 11:25:46 ECO-Origami systemd-udevd[22858]: hidraw1: Process '/usr/bin/iptsd-check-device --quiet /dev/hidraw1' terminated by signal ABRT.
أبريل 28 11:25:46 ECO-Origami systemd-udevd[22858]: hidraw1: Failed to wait for spawned command '/usr/bin/iptsd-check-device --quiet /dev/hidraw1': Input/output error
```

StollD, 2023-04-28:
Please post your full dmesg after a fresh boot, without any suspends.

ECO1AI, 2023-04-28:
here is `journalctl -b` after rebooting:

```
13:28:49 ECO-Origami kernel: microcode: microcode updated early to revision 0xf0, date = 2021-11-12
أبريل 28 13:28:49 ECO-Origami kernel: Linux version 6.2.12-surface (root@3b346b1b4b1e) (gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0, GNU ld (GNU Binutils for Ubuntu) 2.>
أبريل 28 13:28:49 ECO-Origami kernel: Command line: BOOT_IMAGE=/@/boot/vmlinuz-6.2.12-surface root=UUID=a3ae7ea3-8b03-4aa8-a2bc-1b48ab26e499 ro rootflags=subvol=@ qui>
أبريل 28 13:28:49 ECO-Origami kernel: KERNEL supported cpus:
أبريل 28 13:28:49 ECO-Origami kernel:   Intel GenuineIntel
أبريل 28 13:28:49 ECO-Origami kernel:   AMD AuthenticAMD
أبريل 28 13:28:49 ECO-Origami kernel:   Hygon HygonGenuine
أبريل 28 13:28:49 ECO-Origami kernel:   Centaur CentaurHauls
أبريل 28 13:28:49 ECO-Origami kernel:   zhaoxin   Shanghai  
أبريل 28 13:28:49 ECO-Origami kernel: x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
أبريل 28 13:28:49 ECO-Origami kernel: x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
أبريل 28 13:28:49 ECO-Origami kernel: x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
أبريل 28 13:28:49 ECO-Origami kernel: x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
أبريل 28 13:28:49 ECO-Origami kernel: x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
أبريل 28 13:28:49 ECO-Origami kernel: x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
أبريل 28 13:28:49 ECO-Origami kernel: x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
أبريل 28 13:28:49 ECO-Origami kernel: x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
أبريل 28 13:28:49 ECO-Origami kernel: x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
أبريل 28 13:28:49 ECO-Origami kernel: signal: max sigframe size: 2032
lines 1-19
```

qzed, 2023-04-28:
This is only the first part of the log. Please post the full log (e.g. by redirecting it to a file via `journalctl -b > journalctl.log` and submitting that here).

ECO1AI, 2023-04-28:
ops
sorry
my bad 
[journalctl.log](https://github.com/linux-surface/intel-precise-touch/files/11352846/journalctl.log)
there you go

StollD, 2023-04-28:
Ok, so this is interesting. iptsd, the component that enables multitouch and stylus support, doesn't launch at all. 

The log shows that the kernel registers two `hidraw0` devices, one from IPTS and one from HID over I2C (which iirc is something with the EC?). Could you run `sudo iptsd-check-device /dev/hidraw0` and post the output? The output of `ls -la /dev/hidraw*` could also be useful.

The second issue is that iptsd-check-device seems to crash when it is run against the hidraw device of your bluetooth keyboard. I am not really sure why that would even happen, and it shouldn't stop iptsd from launching. But just to be sure, could you disconnect the bluetooth keyboard and check if that makes iptsd launch (simply reboot and check if the stylus works, this should work in the login screen too)?

StollD, 2023-04-28:
It is also possible that I messed up and accidentally brought back https://github.com/linux-surface/linux-surface/issues/1048, since the IPTS device appears two times, once for hid-generic and once for hid-multitouch.

ECO1AI, 2023-04-28:
`sudo iptsd-check-device /dev/hidraw0` output :
```
[14:47:49.025] [info] Opened device 1B96:006A
[14:47:49.026] [error] Failed to parse /etc/iptsd.conf
```
note: i changed the config file to avoid ghost touch, but issue is before the change.

`ls -la /dev/hidraw*` output:
```
crw------- 1 root root 240, 0 أبريل  28 14:45 /dev/hidraw0
crw------- 1 root root 240, 1 أبريل  28 14:46 /dev/hidraw1
```

> Ok, so this is interesting. iptsd, the component that enables multitouch and stylus support, doesn't launch at all.

the weird thing is that touch work normally. 
So, how is touch working but iptsd is not working?

> It is also possible that I messed up and accidentally brought back [linux-surface/linux-surface#1048](https://github.com/linux-surface/linux-surface/issues/1048)

following the link i tried:
`sudo systemctl start $(iptsd-find-service)` that didnt work
iptsd is dead for sure

here is the log after trying to start iptsd manually (bluetooth keyboard is off)
```

أبريل 28 15:02:47 ECO-Origami systemd[1]: Started iptsd@dev-hidraw0.service - Intel Precise Touch & Stylus Daemon.
أبريل 28 15:02:47 ECO-Origami systemd[1]: Stopping iptsd@dev-hidraw0.service - Intel Precise Touch & Stylus Daemon...
أبريل 28 15:02:47 ECO-Origami systemd[1]: iptsd@dev-hidraw0.service: Deactivated successfully.
أبريل 28 15:02:47 ECO-Origami systemd[1]: Stopped iptsd@dev-hidraw0.service - Intel Precise Touch & Stylus Daemon.

```
`which iptsd` returns `which iptsd` which is correct

> But just to be sure, could you disconnect the bluetooth keyboard and check if that makes iptsd launch

no change

i think it is safe to ignore this issue for now(at least)

ECO1AI, 2023-04-28:
here is ipts config file(just in case)
[iptsd.txt](https://github.com/linux-surface/intel-precise-touch/files/11353391/iptsd.txt)

StollD, 2023-04-28:
> the weird thing is that touch work normally.

Singletouch doesn't need iptsd and is handled internally in the kernel driver. You will notice that multitouch won't work, e.g. you cant zoom in Firefox.

> `sudo iptsd-check-device /dev/hidraw0` output :
> here is ipts config file(just in case)

The first line in that file is `/etc/iptsd.conf`, is that also in the original file? Because that makes it invalid, and it would explain why iptsd cannot run. Other than that I cannot spot any issues with the config.

That said, the config should be unneccessary with the latest version of iptsd. There were some large changes that should notably improve the behaviour on SP4. So you could try to just reset the config to the default values.

https://github.com/linux-surface/iptsd/blob/v1.2.0/etc/iptsd.conf

What confuses me a bit is that the log didn't contain any mention of the invalid config file error.

ECO1AI, 2023-04-28:
> The first line in that file is `/etc/iptsd.conf`, is that also in the original file? Because that makes it invalid, and it would explain why iptsd cannot run. Other than that I cannot spot any issues with the config.

that did the trick only for one boot
after rebooting i got this:
```
19:10:51 ECO-Origami kernel: ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: SET_MODE: recv failed: -11
أبريل 28 19:10:51 ECO-Origami kernel: ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Failed to set mode: -11
أبريل 28 19:10:51 ECO-Origami kernel: ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Failed to switch modes: -11
أبريل 28 19:10:59 ECO-Origami bluetoothd[1265]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
أبريل 28 19:11:54 ECO-Origami systemd-udevd[1852]: hidraw1: Process '/usr/bin/iptsd-check-device --quiet /dev/hidraw1' terminated by signal ABRT.
أبريل 28 19:11:54 ECO-Origami systemd-udevd[1852]: hidraw1: Failed to wait for spawned command '/usr/bin/iptsd-check-device --quiet /dev/hidraw1': Input/output error

```
after running `systemctl status $(iptsd-find-service)` i got this:
```
ECO-Origami% systemctl status $(iptsd-find-service)
× iptsd@dev-hidraw0.service - Intel Precise Touch & Stylus Daemon
     Loaded: loaded (/lib/systemd/system/iptsd@.service; static)
     Active: failed (Result: exit-code) since Fri 2023-04-28 19:10:51 +03; 6min ago
   Duration: 59ms
       Docs: https://github.com/linux-surface/iptsd
    Process: 789 ExecStart=/usr/bin/iptsd /dev/hidraw0 (code=exited, status=1/FAILURE)
   Main PID: 789 (code=exited, status=1/FAILURE)
        CPU: 845ms

أبريل 28 19:10:48 ECO-Origami systemd[1]: Started iptsd@dev-hidraw0.service - Intel Precise Touch & Stylus Daemon.
أبريل 28 19:10:48 ECO-Origami iptsd[789]: [19:10:48.927] [info] Connected to device 1B96:006A
أبريل 28 19:10:48 ECO-Origami systemd[1]: Stopping iptsd@dev-hidraw0.service - Intel Precise Touch & Stylus Daemon...
أبريل 28 19:10:51 ECO-Origami iptsd[789]: [19:10:51.986] [error] Resource temporarily unavailable
أبريل 28 19:10:51 ECO-Origami systemd[1]: iptsd@dev-hidraw0.service: Main process exited, code=exited, status=1/FAILURE
أبريل 28 19:10:51 ECO-Origami systemd[1]: iptsd@dev-hidraw0.service: Failed with result 'exit-code'.
أبريل 28 19:10:51 ECO-Origami systemd[1]: Stopped iptsd@dev-hidraw0.service - Intel Precise Touch & Stylus Daemon.

```
and now even touch is not working

> What confuses me a bit is that the log didn't contain any mention of the invalid config file error.

that also confused me.
it seems like there are multiple issues which prevented rising the error

StollD, 2023-04-28:
What happens if you reload the module?

```
$ sudo modprobe -r ipts
$ sudo modprobe ipts
```

At least iptsd is launching now (and crashing the driver) so this seems like the usual cursed IPTS hardware.

ECO1AI, 2023-04-28:
> What happens if you reload the module?
> 
> ```
> $ sudo modprobe -r ipts
> $ sudo modprobe ipts
> ```
> 
> At least iptsd is launching now (and crashing the driver) so this seems like the usual cursed IPTS hardware.

it work now
```
CO-Origami% sudo modprobe -r ipts
[sudo] password for yassen: 
ECO-Origami% sudo modprobe ipts
ECO-Origami% systemctl status $(iptsd-find-service)
● iptsd@dev-hidraw2.service - Intel Precise Touch & Stylus Daemon
     Loaded: loaded (/lib/systemd/system/iptsd@.service; static)
     Active: active (running) since Fri 2023-04-28 19:46:32 +03; 43s ago
       Docs: https://github.com/linux-surface/iptsd
   Main PID: 4070 (iptsd)
      Tasks: 1 (limit: 4468)
     Memory: 1.0M
        CPU: 27ms
     CGroup: /system.slice/system-iptsd.slice/iptsd@dev-hidraw2.service
             └─4070 /usr/bin/iptsd /dev/hidraw2

أبريل 28 19:46:32 ECO-Origami systemd[1]: Started iptsd@dev-hidraw2.service - Intel Precise Touch & Stylus Daemon.
أبريل 28 19:46:32 ECO-Origami iptsd[4070]: [19:46:32.831] [info] Connected to device 1B96:006A
ECO-Origami% 

```

ECO1AI, 2023-04-28:
[dmesg.log](https://github.com/linux-surface/intel-precise-touch/files/11355727/dmesg.log)
[journalctl.log](https://github.com/linux-surface/intel-precise-touch/files/11355741/journalctl.log)
it seems like both files contain a hint, but i dont get it 

these are from a clean boot

StollD, 2023-04-28:
Your hardware is timing out. I have no idea why that happens.

Does reloading the module after booting reliably fix it? Could you also try to reload the module a few times to see if it breaks again?

If this is only reproduceable during boot, debugging will be a lot more annoying.

ECO1AI, 2023-04-28:
> Does reloading the module after booting reliably fix it?

it seems so.

for now i have to  find a way to reload them automatically on boot ( either delaying the module or restarting it).
hopefully next kernel update fixes it.

for now i will keep this issue open for two days :
in this while, i will find a way to automate
and keep testing

ECO1AI, 2023-05-12:
for any some reasons it fixed it self on its own.
now it is working as normal, but:
```
systemd-udevd[358]: event_source: Failed to get device name: No such file or directory

systemd-udevd[2106]: hidraw1: Process '/usr/bin/iptsd-check-device --quiet /dev/hidraw1' terminated by signal ABRT.
systemd-udevd[2106]: hidraw1: Failed to wait for spawned command '/usr/bin/iptsd-check-device --quiet /dev/hidraw1': 
```
these error are in journal, they dont seem to be harmful though.

@StollD @qzed thanks for your help.

closed 2023-05-12 by ECO1AI