linux-surface/intel-precise-touch #25: How can i install this drivers? Help me

SwagSpaad, 2023-09-06:
I have just installed Arch on my SP8. I'm trying to install these drivers, but when i have to patch the kernel i can't find the  "/drivers/misc/mei/hw-me-regs.h" and  "/drivers/misc/mei/pci-me.c" files to apply the patches. Can someone help me?

SwagSpaad, 2023-09-06:
Also tried to find them but that's the output
[image]

qzed, 2023-09-06:
You don't have to patch the kernel yourself. Unless you want to compile and package the kernel from scratch and you're absolutely sure that's what you want to do (in that case, you can get some already patched sources from https://github.com/linux-surface/kernel/ or just pick and apply the patches from https://github.com/linux-surface/linux-surface/tree/master/patches).

For a "normal" install, just follow the instructions at https://github.com/linux-surface/linux-surface/wiki. That describes installing a kernel that's already patched, as well as some other packages that you might need or might be useful.

SwagSpaad, 2023-09-06:
Ok, thought i had to do this to make my stylus work better. Is it normal that it is recognised as a secondary mouse and when i try to use xournal to write the lines sometimes are dotted?

qzed, 2023-09-06:
It being recognized as a mouse depends on the application. I don't remember how Xournal behaves with it, but Xournal++ should be better. Or you could try rnote.

As for the dotted lines: You can try to tweak some values in `/etc/iptsd.conf`. In particular [these](https://github.com/linux-surface/iptsd/blob/7a84cc34e3ccb81268b47426ab9fe0dfef328619/etc/iptsd.conf#L156-L161) (and comment them out). I think you most likely need to increase `ButtonMinMag`.

SwagSpaad, 2023-09-06:
will try that. I will close if it works better, thank you for the quick answer.

SwagSpaad, 2023-09-06:
Ok, tried to edit the values in that file. It seems better if i use the tip of the stylus on the center, but if i use it on the edge it interrupts the line. I attach the photo to explain u better. Is there something i can do? I saw there is a parameter about cone angle, should i edit that?

On the left a line using thecenter of the tip and on the right using by tilting the pen and using it on the edge of the tip
[image]

qzed, 2023-09-06:
The `[Cone]` section is for blocking palm touches when the stylus is in use, so I don't think changing that will help. The `[DFT]` section influences how the raw stylus are interpreted, so I think those are the values you want to experiment with.

SwagSpaad, 2023-09-06:
Ok, where can i find what each parameter on DFT section does? And could the calibration improve this thing?

qzed, 2023-09-06:
I think calibration won't help here. The Gnome one rather breaks things and the IPTSd one is just for touch, not stylus, as far as I know.

Unfortunately the parameters aren't well documented yet and I don't fully understand all of them either. `ButtonMinMag` is essentially a threshold that determines when the side button has been pressed. So if that is too low, it behaves as if the side button is pressed even though it hasn't.

qzed, 2023-09-06:
If you want to, you can try to figure out what the parameters do by checking https://github.com/linux-surface/iptsd/blob/master/src/core/generic/dft.hpp#L227 (search for `m_config`).

SwagSpaad, 2023-09-06:
ok thank you so much

closed 2023-09-06 by SwagSpaad

StollD, 2023-09-06:
I actually have an idea how DFT calibration could work. You capture reports from the stylus, one time with the button pressed and one time without. Then you get the highest magnitude without the button and the lowest magnitude with the button. The new `ButtonMinMag` is the point in the middle between both.

Sadly I am super busy and have no time at all right now ....

qzed, 2023-09-06:
I think ideally average over a couple of measurements, but yeah, that particular value shouldn't be too hard to calibrate for.