linux-surface/intel-precise-touch #26: Getting Multi-touch & Pen support

Theocat321, 2023-10-03:
I'm having problems getting pen support and multi touch working
I can manually start it
`systemctl start $(iptsd-find-service)
`
I get the status by running

`systemctl status $(iptsd-find-service)
`

iptsd@dev-hidraw1.service - Intel Precise Touch & Stylus Daemon 
     Loaded: loaded (/lib/systemd/system/iptsd@.service; static)
     Active: inactive (dead)
       Docs: https://github.com/linux-surface/iptsd

Then Logs...
Oct 03 15:12:37 adam-Surface-Pro-7 systemd[1]: Started Intel Precise Touch & Stylus Daemon.
Oct 03 15:12:37 adam-Surface-Pro-7 systemd[1]: Stopping Intel Precise Touch & Stylus Daemo>
Oct 03 15:12:37 adam-Surface-Pro-7 systemd[1]: iptsd@dev-hidraw1.service: Deactivated succ>
Oct 03 15:12:37 adam-Surface-Pro-7 systemd[1]: Stopped Intel Precise Touch & Stylus Daemon.

Running `which iptsd`
Give me `/usr/bin/iptsd`

Any pointers/ help would be apprecitated

Theocat321, 2023-10-03:
> Please post your dmesg.

Anything specific on it or the whole thing?

StollD, 2023-10-03:
The whole thing

Theocat321, 2023-10-03:
[    0.000000] microcode: updated early: 0xb0 -> 0xbc, date = 2023-02-26
[    0.000000] Linux version 6.5.5-surface (root@81b3f02ee08b) (gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1 SMP PREEMPT_DYNAMIC Tue Oct  3 02:37:09 UTC 2023
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.5.5-surface root=UUID=db8e4fd1-3838-4ee3-94fc-74bbd79dda54 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/split lock detection: #AC: crashing the kernel on kernel split_locks and warning on user-space split_locks
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000079e1ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000079e20000-0x0000000079e20fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000079e21000-0x0000000079e21fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000079e22000-0x0000000079e22fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000079e23000-0x000000007b675fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007b676000-0x000000007ba75fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007ba76000-0x000000007ba8efff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ba8f000-0x000000007bb89fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007bb8a000-0x000000007bbb3fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007bbb4000-0x000000007bbfefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007bbff000-0x000000007bbfffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007bc00000-0x00000000953fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe010000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000046abfffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.7 by MSFT
[    0.000000] efi: ACPI=0x7bbfe000 ACPI 2.0=0x7bbfe014 TPMFinalLog=0x7bbac000 SMBIOS=0x7baff000 SMBIOS 3.0=0x7bafd000 MEMATTR=0x77b1d018 ESRT=0x78f0bb98 MOKvar=0x7bafa000 RNG=0x7bbb7018 TPMEventLog=0x77557018 
[    0.000000] random: crng init done
[    0.000000] efi: Not removing mem186: MMIO range=[0xfe010000-0xfe010fff] (4KB) from e820 map
[    0.000000] SMBIOS 3.3.0 present.
[    0.000000] DMI: Microsoft Corporation Surface Pro 7/Surface Pro 7, BIOS 15.11.140 07/08/2022
[    0.000000] tsc: Detected 1500.000 MHz processor
[    0.000000] tsc: Detected 1497.600 MHz TSC
[    0.001177] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001183] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001199] last_pfn = 0x46ac00 max_arch_pfn = 0x400000000
[    0.001208] MTRR map: 5 entries (3 fixed + 2 variable; max 23), built from 10 variable MTRRs
[    0.001212] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.002117] last_pfn = 0x7bc00 max_arch_pfn = 0x400000000
[    0.033219] esrt: Reserving ESRT space from 0x0000000078f0bb98 to 0x0000000078f0bc70.
[    0.033232] e820: update [mem 0x78f0b000-0x78f0bfff] usable ==> reserved
[    0.033284] Using GB pages for direct mapping
[    0.033770] Secure boot disabled
[    0.033771] RAMDISK: [mem 0x6956f000-0x6d4e4fff]
[    0.033781] ACPI: Early table checksum verification disabled
[    0.033787] ACPI: RSDP 0x000000007BBFE014 000024 (v02 MSFT  )
[    0.033794] ACPI: XSDT 0x000000007BBFD0E8 0000BC (v01 MSFT   MSFT     00000002 MSFT 20160422)
[    0.033805] ACPI: FACP 0x000000007BBE1000 000114 (v06 MSFT   MSFT     00000002 MSFT 20160422)
[    0.033815] ACPI: DSDT 0x000000007BBC0000 01C11E (v02 MSFT   MSFT     00000002 MSFT 20160422)
[    0.033821] ACPI: FACS 0x000000007BB92000 000040
[    0.033827] ACPI: WSMT 0x000000007BBFC000 000028 (v01 MSFT   MSFT     00000002 MSFT 20160422)
[    0.033832] ACPI: SSDT 0x000000007BBFA000 001B61 (v02 CpuRef CpuSsdt  00003000 INTL 20181003)
[    0.033839] ACPI: MSDM 0x000000007BBF9000 000055 (v01 MSFT            00000001 MSFT 00000001)
[    0.033845] ACPI: SSDT 0x000000007BBF2000 006230 (v02 MSFT   DptfTabl 00001000 INTL 20181003)
[    0.033851] ACPI: SSDT 0x000000007BBEE000 0033E3 (v02 SaSsdt SaSsdt   00003000 INTL 20181003)
[    0.033857] ACPI: SSDT 0x000000007BBE2000 00B27A (v02 INTEL  TcssSsdt 00001000 INTL 20181003)
[    0.033863] ACPI: HPET 0x000000007BBE0000 000038 (v01 MSFT   MSFT     00000002 MSFT 20160422)
[    0.033869] ACPI: LPIT 0x000000007BBDF000 000094 (v01 MSFT   MSFT     00000002 MSFT 20160422)
[    0.033875] ACPI: APIC 0x000000007BBDE000 00012C (v04 MSFT   MSFT     00000002 MSFT 20160422)
[    0.033881] ACPI: MCFG 0x000000007BBDD000 00003C (v01 MSFT   MSFT     00000002 MSFT 20160422)
[    0.033887] ACPI: SSDT 0x000000007BBBF000 000DB4 (v02 MSFT   RTD3_JL  00001000 INTL 20181003)
[    0.033893] ACPI: SSDT 0x000000007BBBE000 0009E3 (v02 MSFT   xh_jl000 00000000 INTL 20181003)
[    0.033899] ACPI: NHLT 0x000000007BBBD000 00002D (v00 MSFT   MSFT     00000002 MSFT 20160422)
[    0.033904] ACPI: SSDT 0x000000007BBBC000 000574 (v02 MSFT   Tpm2Tabl 00001000 INTL 20181003)
[    0.033910] ACPI: TPM2 0x000000007BBBB000 000034 (v03 MSFT   MSFT     00000002 MSFT 20160422)
[... cut ...]
[   28.426411] audit: type=1400 audit(1696360924.211:144): apparmor="DENIED" operation="open" class="file" profile="snap.bluemail.bluemail" name="/sys/devices/virtual/dmi/id/product_name" pid=4672 comm="cat" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[   28.436021] audit: type=1400 audit(1696360924.219:145): apparmor="DENIED" operation="open" class="file" profile="snap.bluemail.bluemail" name="/sys/devices/virtual/dmi/id/product_version" pid=4675 comm="cat" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0

Theocat321, 2023-10-03:
> The whole thing

[    0.000000] microcode: updated early: 0xb0 -> 0xbc, date = 2023-02-26
[    0.000000] Linux version 6.5.5-surface (root@81b3f02ee08b) (gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1 SMP PREEMPT_DYNAMIC Tue Oct  3 02:37:09 UTC 2023
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.5.5-surface root=UUID=db8e4fd1-3838-4ee3-94fc-74bbd79dda54 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] x86/split lock detection: #AC: crashing the kernel on kernel split_locks and warning on user-space split_locks
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000079e1ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000079e20000-0x0000000079e20fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000079e21000-0x0000000079e21fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000079e22000-0x0000000079e22fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000079e23000-0x000000007b675fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007b676000-0x000000007ba75fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007ba76000-0x000000007ba8efff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ba8f000-0x000000007bb89fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007bb8a000-0x000000007bbb3fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007bbb4000-0x000000007bbfefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007bbff000-0x000000007bbfffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007bc00000-0x00000000953fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe010000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000046abfffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.7 by MSFT
[    0.000000] efi: ACPI=0x7bbfe000 ACPI 2.0=0x7bbfe014 TPMFinalLog=0x7bbac000 SMBIOS=0x7baff000 SMBIOS 3.0=0x7bafd000 MEMATTR=0x77b1d018 ESRT=0x78f0bb98 MOKvar=0x7bafa000 RNG=0x7bbb7018 TPMEventLog=0x77557018 
[    0.000000] random: crng init done
[    0.000000] efi: Not removing mem186: MMIO range=[0xfe010000-0xfe010fff] (4KB) from e820 map
[    0.000000] SMBIOS 3.3.0 present.
[    0.000000] DMI: Microsoft Corporation Surface Pro 7/Surface Pro 7, BIOS 15.11.140 07/08/2022
[    0.000000] tsc: Detected 1500.000 MHz processor
[    0.000000] tsc: Detected 1497.600 MHz TSC
[    0.001177] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001183] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001199] last_pfn = 0x46ac00 max_arch_pfn = 0x400000000
[    0.001208] MTRR map: 5 entries (3 fixed + 2 variable; max 23), built from 10 variable MTRRs
[    0.001212] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.002117] last_pfn = 0x7bc00 max_arch_pfn = 0x400000000
[    0.033219] esrt: Reserving ESRT space from 0x0000000078f0bb98 to 0x0000000078f0bc70.
[    0.033232] e820: update [mem 0x78f0b000-0x78f0bfff] usable ==> reserved
[    0.033284] Using GB pages for direct mapping
[    0.033770] Secure boot disabled
[    0.033771] RAMDISK: [mem 0x6956f000-0x6d4e4fff]
[    0.033781] ACPI: Early table checksum verification disabled
[    0.033787] ACPI: RSDP 0x000000007BBFE014 000024 (v02 MSFT  )
[    0.033794] ACPI: XSDT 0x000000007BBFD0E8 0000BC (v01 MSFT   MSFT     00000002 MSFT 20160422)
[    0.033805] ACPI: FACP 0x000000007BBE1000 000114 (v06 MSFT   MSFT     00000002 MSFT 20160422)
[    0.033815] ACPI: DSDT 0x000000007BBC0000 01C11E (v02 MSFT   MSFT     00000002 MSFT 20160422)
[    0.033821] ACPI: FACS 0x000000007BB92000 000040
[    0.033827] ACPI: WSMT 0x000000007BBFC000 000028 (v01 MSFT   MSFT     00000002 MSFT 20160422)
[    0.033832] ACPI: SSDT 0x000000007BBFA000 001B61 (v02 CpuRef CpuSsdt  00003000 INTL 20181003)
[    0.033839] ACPI: MSDM 0x000000007BBF9000 000055 (v01 MSFT            00000001 MSFT 00000001)
[    0.033845] ACPI: SSDT 0x000000007BBF2000 006230 (v02 MSFT   DptfTabl 00001000 INTL 20181003)
[    0.033851] ACPI: SSDT 0x000000007BBEE000 0033E3 (v02 SaSsdt SaSsdt   00003000 INTL 20181003)
[    0.033857] ACPI: SSDT 0x000000007BBE2000 00B27A (v02 INTEL  TcssSsdt 00001000 INTL 20181003)
[    0.033863] ACPI: HPET 0x000000007BBE0000 000038 (v01 MSFT   MSFT     00000002 MSFT 20160422)
[    0.033869] ACPI: LPIT 0x000000007BBDF000 000094 (v01 MSFT   MSFT     00000002 MSFT 20160422)
[    0.033875] ACPI: APIC 0x000000007BBDE000 00012C (v04 MSFT   MSFT     00000002 MSFT 20160422)
[    0.033881] ACPI: MCFG 0x000000007BBDD000 00003C (v01 MSFT   MSFT     00000002 MSFT 20160422)
[    0.033887] ACPI: SSDT 0x000000007BBBF000 000DB4 (v02 MSFT   RTD3_JL  00001000 INTL 20181003)
[    0.033893] ACPI: SSDT 0x000000007BBBE000 0009E3 (v02 MSFT   xh_jl000 00000000 INTL 20181003)
[    0.033899] ACPI: NHLT 0x000000007BBBD000 00002D (v00 MSFT   MSFT     00000002 MSFT 20160422)
[    0.033904] ACPI: SSDT 0x000000007BBBC000 000574 (v02 MSFT   Tpm2Tabl 00001000 INTL 20181003)
[... 1037 lines cut ...]
[... cut ...]
[   28.426411] audit: type=1400 audit(1696360924.211:144): apparmor="DENIED" operation="open" class="file" profile="snap.bluemail.bluemail" name="/sys/devices/virtual/dmi/id/product_name" pid=4672 comm="cat" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[   28.436021] audit: type=1400 audit(1696360924.219:145): apparmor="DENIED" operation="open" class="file" profile="snap.bluemail.bluemail" name="/sys/devices/virtual/dmi/id/product_version" pid=4675 comm="cat" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0

StollD, 2023-10-05:
> [ 5.190929] ipts: loading out-of-tree module taints kernel.
[ 5.190934] ipts: module verification failed: signature and/or required key missing - tainting kernel

Why are you using an out of tree / DKMS build of the driver? The surface builds include the latest (stable) version.

> [ 11.659611] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: GET_FEATURES timed out!

That is your issue. The hardware times out when iptsd tries to read device information from it.

To my knowledge this error doesn't happen on the version of the driver that is included in the surface kernel. It was triggered - for a reason that I can't explain - by some recent changes I did: https://github.com/linux-surface/intel-precise-touch/compare/4b5c013d0140c47e4a009adf9a053a45bb0532a3...master.

The fix was to massively increase the time that the driver will wait for a response: https://github.com/linux-surface/intel-precise-touch/commit/4802f822199e2c185228e416582544f731040a90. From your log I can see that your build doesn't have this change applied, because it times out 11 seconds after boot. It can't have waited for 30 seconds already.

So please try to either remove your custom version of the module and use the one included in the surface kernel, or install the latest version from master (unless you already did that)

Theocat321, 2023-10-05:
I was originally on an unstable version of the kernal as I wanted the multi touch features. Since then, it has been moved into the main branch. I have tried to install the latest kernal and it hasn't changed anything. To be honest not sure how to remove my custom version!!

StollD, 2023-10-05:
`sudo dkms remove "ipts/1.0.0" --force --all`, then reinstall the latest kernel

Theocat321, 2023-10-05:
Thank you so much!! That has all worked. Writing notes in Uni was getting annoying but the pen is back!!

StollD, 2023-10-09:
Thats a different issue. In your case, it seems that the hardware takes too long to respond to one of the commands so the driver timeouts.

You could try if the latest version of the module fixes it:

```
$ git clone https://github.com/linux-surface/intel-precise-touch
$ cd intel-precise-touch
$ sudo make dkms-install
```

And then reboot (you have to disable secureboot, or setup DKMS for automatic module signing).

If that doesn't fix it, you could try to increase the timeout. In this line, edit `1 * MSEC_PER_SEC` to maybe `5 * MSEC_PER_SEC` (5 seconds instead of one): https://github.com/linux-surface/intel-precise-touch/blob/08e4dcec41c454387fb1c52a26a1cacbd10ffabb/src/mei.h#L63

Then run

```
$ sudo make dkms-uninstall
$ sudo make dkms-install
```

digitables, 2023-10-19:
Hi, I'm getting an issue where multitouch is occurring when it shouldn't. I'm trying to draw in xournal or rnote and my palm is being interpreted as a second input, causing the device to scroll. 
OS: Fedora 38
Kernel: 6.4.12-1.surface.fc38.x86_64 or  6.5.5-2.surface.fc38.x86_64 kernel (the problem persists with both of them)

digitables, 2023-10-19:
Dmesg:
```
[    0.000000] microcode: updated early: 0xb0 -> 0xbc, date = 2023-02-26
[    0.000000] Linux version 6.4.12-1.surface.fc38.x86_64 (root@cc553d3fd00f) (gcc (GCC) 13.2.1 20230728 (Red Hat 13.2.1-1), GNU ld version 2.39-9.fc38) #1 SMP PREEMPT_DYNAMIC Fri Aug 25 21:30:59 UTC 2023
[    0.000000] Command line: BOOT_IMAGE=(hd0,gpt2)/vmlinuz-6.4.12-1.surface.fc38.x86_64 root=UUID=d7e2cb9e-5573-436a-ad6c-4fb3d0c23042 ro resume=UUID=29d97648-f509-462d-ab94-5693d0c71dd7 rd.luks.uuid=luks-063da1c7-66d9-4aad-8fe5-5d317bc8ae67 rhgb quiet
[    0.000000] x86/split lock detection: #AC: crashing the kernel on kernel split_locks and warning on user-space split_locks
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000079e1ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000079e20000-0x0000000079e20fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000079e21000-0x0000000079e21fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000079e22000-0x0000000079e22fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000079e23000-0x000000007b66efff] usable
[    0.000000] BIOS-e820: [mem 0x000000007b66f000-0x000000007ba6efff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007ba6f000-0x000000007ba87fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ba88000-0x000000007bb89fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007bb8a000-0x000000007bbb3fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007bbb4000-0x000000007bbfefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007bbff000-0x000000007bbfffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007bc00000-0x00000000953fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe010000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000046abfffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.7 by MSFT
[    0.000000] efi: ACPI=0x7bbfe000 ACPI 2.0=0x7bbfe014 TPMFinalLog=0x7bbac000 SMBIOS=0x7baff000 SMBIOS 3.0=0x7bafd000 MEMATTR=0x77a92018 ESRT=0x77b7ae98 MOKvar=0x7bafb000 RNG=0x7bbb7018 TPMEventLog=0x77414018 
[    0.000000] random: crng init done
[    0.000000] efi: Not removing mem188: MMIO range=[0xfe010000-0xfe010fff] (4KB) from e820 map
[    0.000000] secureboot: Secure boot enabled
[    0.000000] Kernel is locked down from EFI Secure Boot mode; see man kernel_lockdown.7
[    0.000000] SMBIOS 3.3.0 present.
[    0.000000] DMI: Microsoft Corporation Surface Pro 7/Surface Pro 7, BIOS 15.11.140 07/08/2022
[    0.000000] tsc: Detected 1500.000 MHz processor
[    0.000000] tsc: Detected 1497.600 MHz TSC
[    0.001157] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001163] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001179] last_pfn = 0x46ac00 max_arch_pfn = 0x400000000
[    0.001187] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.002001] last_pfn = 0x7bc00 max_arch_pfn = 0x400000000
[    0.031247] esrt: Reserving ESRT space from 0x0000000077b7ae98 to 0x0000000077b7af98.
[    0.031259] e820: update [mem 0x77b7a000-0x77b7afff] usable ==> reserved
[    0.031314] Using GB pages for direct mapping
[    0.032368] secureboot: Secure boot enabled
[    0.032369] RAMDISK: [mem 0x6e4aa000-0x707c6fff]
[    0.032378] ACPI: Early table checksum verification disabled
[    0.032382] ACPI: RSDP 0x000000007BBFE014 000024 (v02 MSFT  )
[    0.032389] ACPI: XSDT 0x000000007BBFD0E8 0000BC (v01 MSFT   MSFT     00000002 MSFT 20160422)
[    0.032399] ACPI: FACP 0x000000007BBE1000 000114 (v06 MSFT   MSFT     00000002 MSFT 20160422)
[    0.032408] ACPI: DSDT 0x000000007BBC0000 01C11E (v02 MSFT   MSFT     00000002 MSFT 20160422)
[    0.032413] ACPI: FACS 0x000000007BB92000 000040
[    0.032418] ACPI: WSMT 0x000000007BBFC000 000028 (v01 MSFT   MSFT     00000002 MSFT 20160422)
[    0.032423] ACPI: SSDT 0x000000007BBFA000 001B61 (v02 CpuRef CpuSsdt  00003000 INTL 20181003)
[    0.032428] ACPI: MSDM 0x000000007BBF9000 000055 (v01 MSFT            00000001 MSFT 00000001)
[    0.032434] ACPI: SSDT 0x000000007BBF2000 006230 (v02 MSFT   DptfTabl 00001000 INTL 20181003)
[    0.032439] ACPI: SSDT 0x000000007BBEE000 0033E3 (v02 SaSsdt SaSsdt   00003000 INTL 20181003)
[    0.032444] ACPI: SSDT 0x000000007BBE2000 00B27A (v02 INTEL  TcssSsdt 00001000 INTL 20181003)
[    0.032449] ACPI: HPET 0x000000007BBE0000 000038 (v01 MSFT   MSFT     00000002 MSFT 20160422)
[    0.032454] ACPI: LPIT 0x000000007BBDF000 000094 (v01 MSFT   MSFT     00000002 MSFT 20160422)
[    0.032459] ACPI: APIC 0x000000007BBDE000 00012C (v04 MSFT   MSFT     00000002 MSFT 20160422)
[    0.032464] ACPI: MCFG 0x000000007BBDD000 00003C (v01 MSFT   MSFT     00000002 MSFT 20160422)
[    0.032469] ACPI: SSDT 0x000000007BBBF000 000DB4 (v02 MSFT   RTD3_JL  00001000 INTL 20181003)
[    0.032474] ACPI: SSDT 0x000000007BBBE000 0009E3 (v02 MSFT   xh_jl000 00000000 INTL 20181003)
[    0.032480] ACPI: NHLT 0x000000007BBBD000 00002D (v00 MSFT   MSFT     00000002 MSFT 20160422)
[    0.032485] ACPI: SSDT 0x000000007BBBC000 000574 (v02 MSFT   Tpm2Tabl 00001000 INTL 20181003)
[    0.032490] ACPI: TPM2 0x000000007BBBB000 000034 (v03 MSFT   MSFT     00000002 MSFT 20160422)
[... cut ...]
[  903.171183] input: Microsoft Surface Type Cover Tablet Mode Switch as /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:045E:09C0.0007/input/input122
[  903.171254] input: Microsoft Surface Type Cover Tablet Mode Switch as /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:045E:09C0.0007/input/input123
[  903.171324] input: Microsoft Surface Type Cover Tablet Mode Switch as /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:045E:09C0.0007/input/input124
[  903.171539] hid-multitouch 0003:045E:09C0.0007: input,hiddev96,hidraw0: USB HID v1.11 Keyboard [Microsoft Surface Type Cover] on usb-0000:00:14.0-8/input0
```

StollD, 2023-10-19:
Given that the problem of the OP (the touchscreen not working at all) isn't similar to your issue, and since you already found an issue that describes the problem you have perfectly (https://github.com/linux-surface/linux-surface/issues/1273) I am not sure why you are posting here.

I would understand opening a new issue if you weren't sure if the one on linux-surface gets enough attention. But people, please don't try to take over any random issue just because its open and has a somewhat generic title.

Closing this now.

closed 2023-10-19 by StollD