linux-surface/intel-precise-touch #27: Slim Pen 2 Haptic Feedback

MediciPrime, 2023-11-11:
I am not sure where I could pose this question so please delete if it's not allowed here. 

Does anyone know of the current state of the slim pen 2's haptic feedback support?

I am currently running Fedora 39 on a Surface 9 Pro and it's working very well (better than windows). The only thing that isn't working is the haptic feedback of the slim pen 2. 

Is there any way to turn on the haptic feedback whenever the slim pen 2 touches the screen regardless what app is running?

qzed, 2023-12-09:
There is https://github.com/linux-surface/linux-surface/issues/1066. Someone apparently got it to vibrate.

antoniovazquezblanco, 2026-11-07:
As per https://github.com/linux-surface/kernel/pull/178#issuecomment-6447636724, it seems that the correct place to send PRs and bug reports is now https://github.com/linux-surface/kernel.

This repo should be archived or updated by the maintainers!