linux-surface/intel-precise-touch #33: ipts suspend failure

danielzgtg, 2024-12-25:
My SP7 randomly failed to suspend today and the screen was, instead of turning off, frozen for 20 seconds.

<details>
<summary>dmesg</summary>

```
[1262705.390517] PM: suspend exit
[1262705.428093] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_ops [i915])
[1262705.428751] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Starting IPTS
[1262705.430124] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: IPTS EDS Version: 2
[1262705.439169] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: IPTS running in poll mode
[1262705.440603] input: IPTS 045E:099F Touchscreen as /devices/pci0000:00/0000:00:16.4/0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04/0000:045E:099F.0031/input/input498
[1262705.440694] input: IPTS 045E:099F as /devices/pci0000:00/0000:00:16.4/0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04/0000:045E:099F.0031/input/input499
[1262705.440741] input: IPTS 045E:099F as /devices/pci0000:00/0000:00:16.4/0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04/0000:045E:099F.0031/input/input500
[1262705.440781] input: IPTS 045E:099F as /devices/pci0000:00/0000:00:16.4/0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04/0000:045E:099F.0031/input/input501
[1262705.440824] input: IPTS 045E:099F as /devices/pci0000:00/0000:00:16.4/0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04/0000:045E:099F.0031/input/input502
[1262705.440868] input: IPTS 045E:099F Stylus as /devices/pci0000:00/0000:00:16.4/0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04/0000:045E:099F.0031/input/input503
[1262705.441027] input: IPTS 045E:099F as /devices/pci0000:00/0000:00:16.4/0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04/0000:045E:099F.0031/input/input504
[1262705.441084] input: IPTS 045E:099F as /devices/pci0000:00/0000:00:16.4/0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04/0000:045E:099F.0031/input/input505
[1262705.639612] Oops: general protection fault, probably for non-canonical address 0xdead000000000108: 0000 [#1] PREEMPT SMP NOPTI
[1262705.639618] hid-generic 0000:045E:099F.0031: input,hidraw2: <UNKNOWN> HID v0.00 Device [IPTS 045E:099F] on 
[1262705.639621] CPU: 3 PID: 483206 Comm: ipts_poll Not tainted 6.10.5-surface-1 #1
[1262705.639625] Hardware name: Microsoft Corporation Surface Pro 7/Surface Pro 7, BIOS 20.103.140 08/16/2024
[1262705.639627] RIP: 0010:ipts_mei_search+0x7b/0x140 [ipts]
[1262705.639632] Code: c4 0f 84 c8 00 00 00 45 39 6c 24 10 75 ec 4c 89 f7 e8 89 46 ba e2 4c 89 f7 e8 21 9e a3 e3 49 8b 14 24 49 8b 44 24 08 4c 89 f7 <48> 89 42 08 48 89 10 48 b8 00 01 00 00 00 00 ad de 49 89 04 24 48
[1262705.639634] RSP: 0000:ffffa59ca7087d20 EFLAGS: 00010286
[1262705.639637] RAX: dead000000000122 RBX: ffffa59ca7087e00 RCX: 0000000000000000
[1262705.639639] RDX: dead000000000100 RSI: 0000000000000000 RDI: ffff9957cffc2860
[1262705.639641] RBP: ffffa59ca7087d48 R08: ffffa59ca7087d70 R09: ffff9957cffc2850
[1262705.639642] R10: ffff9957cffc2850 R11: 0000000000000000 R12: ffff995a4368cce8
[1262705.639644] R13: 0000000080000006 R14: ffff9957cffc2860 R15: ffff9957cffc2830
[1262705.639646] FS:  0000000000000000(0000) GS:ffff995b1ab80000(0000) knlGS:0000000000000000
[1262705.639648] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[1262705.639649] CR2: 0000702512cc2030 CR3: 00000002222bc001 CR4: 0000000000770ef0
[1262705.639651] PKRU: 55555554
[1262705.639653] Call Trace:
[1262705.639655]  <TASK>
[1262705.639657]  ? show_regs+0x67/0x70
[1262705.639663]  ? __die_body+0x20/0x70
[1262705.639666]  ? die_addr+0x3e/0x60
[1262705.639668]  ? exc_general_protection+0x1ea/0x410
[1262705.639674]  ? asm_exc_general_protection+0x27/0x30
[1262705.639678]  ? ipts_mei_search+0x7b/0x140 [ipts]
[1262705.639681]  ipts_mei_recv+0xdb/0x150 [ipts]
[1262705.639684]  ? __pfx_autoremove_wake_function+0x10/0x10
[1262705.639689]  ipts_cmd_recv_timeout+0x30/0xa0 [ipts]
[1262705.639692]  ipts_control_send_feedback+0x7c/0x110 [ipts]
[1262705.639696]  ipts_receiver_poll_loop+0x76/0x1f0 [ipts]
[1262705.639700]  ? __pfx_ipts_thread_runner+0x10/0x10 [ipts]
[1262705.639704]  ipts_thread_runner+0x1f/0x50 [ipts]
[1262705.639707]  kthread+0xe4/0x120
[1262705.639710]  ? __pfx_kthread+0x10/0x10
[1262705.639712]  ret_from_fork+0x39/0x60
[1262705.639715]  ? __pfx_kthread+0x10/0x10
[1262705.639717]  ret_from_fork_asm+0x1a/0x30
[1262705.639722]  </TASK>
[1262705.639723] Modules linked in: rndis_host xt_conntrack nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack_netlink nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xfrm_user xfrm_algo xt_addrtype nft_compat nf_tables libcrc32c br_netfilter bridge stp llc snd_seq_dummy snd_hrtimer rfcomm cmac algif_hash algif_skcipher af_alg bnep overlay snd_sof_pci_intel_icl snd_sof_pci_intel_cnl cmdlinepart snd_sof_intel_hda_generic spi_nor soundwire_intel soundwire_cadence mtd snd_sof_intel_hda_common snd_sof_intel_hda_mlink snd_sof_intel_hda snd_sof_pci snd_sof_xtensa_dsp mei_hdcp mei_pxp ipts snd_sof snd_sof_utils snd_soc_hdac_hda snd_soc_acpi_intel_match soundwire_generic_allocation snd_soc_acpi intel_rapl_msr soundwire_bus snd_soc_avs snd_soc_hda_codec snd_hda_ext_core gpio_keys snd_soc_core surface_gpe snd_hda_codec_realtek snd_compress snd_hda_codec_generic snd_hda_scodec_component ac97_bus snd_pcm_dmaengine input_leds snd_hda_intel snd_intel_dspcfg snd_intel_sdw_acpi mac_hid snd_hda_codec snd_hda_core snd_hwdep joydev
[1262705.639776]  intel_uncore_frequency snd_pcm intel_uncore_frequency_common snd_seq_midi snd_seq_midi_event intel_tcc_cooling x86_pkg_temp_thermal snd_rawmidi intel_powerclamp coretemp snd_seq kvm_intel snd_seq_device iwlmvm kvm binfmt_misc snd_timer spi_intel_pci surface_platform_profile snd rapl mac80211 intel_cstate intel_wmi_thunderbolt surface_battery platform_profile surface_charger libarc4 soundcore spi_intel i915 mei_me mei iwlwifi btusb btrtl processor_thermal_device_pci_legacy btintel hid_sensor_gyro_3d hid_sensor_rotation nls_iso8859_1 processor_thermal_device btbcm btmtk processor_thermal_wt_hint drm_buddy processor_thermal_rfim ttm hid_sensor_accel_3d cfg80211 bluetooth processor_thermal_rapl hid_sensor_trigger intel_rapl_common industrialio_triggered_buffer hid_sensor_iio_common drm_display_helper processor_thermal_wt_req cec processor_thermal_power_floor rc_core processor_thermal_mbox i2c_algo_bit intel_skl_int3472_tps68470 intel_soc_dts_iosf tps68470_regulator ov5693 ov8865 v4l2_cci ov7251 apds9960
[1262705.639829]  v4l2_fwnode clk_tps68470 kfifo_buf industrialio v4l2_async videodev mc intel_skl_int3472_discrete surfacepro3_button
[1262705.639836] ov8865 i2c-INT347A:00: supply dvdd not found, using dummy regulator
[1262705.639837]  soc_button_array
[1262705.639839]  int3403_thermal intel_pmc_core int340x_thermal_zone pinctrl_sunrisepoint surface_acpi_notify intel_vsec pmt_telemetry int3400_thermal acpi_pad dptf_power pmt_class acpi_thermal_rel acpi_tad parport_pc ppdev lp parport nvme_fabrics efi_pstore nfnetlink dmi_sysfs ip_tables x_tables
[1262705.639856] input: IPTS Touch as /devices/virtual/input/input506
[1262705.639859]  autofs4 surface_aggregator_hub surface_hid surface_hid_core hid_sensor_hub intel_ishtp_hid cdc_ether usbnet r8152 mii hid_multitouch hid_generic usbhid hid 8250_dw crct10dif_pclmul crc32_pclmul polyval_clmulni polyval_generic
[1262705.639872] ov8865 i2c-INT347A:00: supply dovdd not found, using dummy regulator
[1262705.639874]  ghash_clmulni_intel nvme sha256_ssse3 sha1_ssse3 nvme_core nvme_auth intel_lpss_pci xhci_pci intel_ish_ipc intel_lpss surface_aggregator_registry video idma64 xhci_pci_renesas intel_ishtp surface_aggregator crc_itu_t wmi pinctrl_icelake aesni_intel crypto_simd cryptd [last unloaded: i2c_hid]
[1262705.639893] ---[ end trace 0000000000000000 ]---
[1262705.639990] input: IPTS Stylus as /devices/virtual/input/input507
[1262705.729726] RIP: 0010:ipts_mei_search+0x7b/0x140 [ipts]
[1262705.729739] Code: c4 0f 84 c8 00 00 00 45 39 6c 24 10 75 ec 4c 89 f7 e8 89 46 ba e2 4c 89 f7 e8 21 9e a3 e3 49 8b 14 24 49 8b 44 24 08 4c 89 f7 <48> 89 42 08 48 89 10 48 b8 00 01 00 00 00 00 ad de 49 89 04 24 48
[... 85 lines cut ...]
[1262804.268288]  do_syscall_64+0x7c/0x160
[1262804.268297]  ? __handle_mm_fault+0x78f/0xe30
[1262804.268310]  ? __count_memcg_events+0x59/0xf0
[1262804.268319]  ? handle_mm_fault+0x22e/0x330
[1262804.268327]  ? do_user_addr_fault+0x5d3/0x850
[1262804.268337]  ? irqentry_exit_to_user_mode+0x73/0x220
[1262804.268350]  ? irqentry_exit+0x3b/0x50
[1262804.268360]  ? clear_bhb_loop+0x15/0x70
[... cut ...]
[1262804.268418] RDX: 000061ed517f6ee0 RSI: 00000000c0024806 RDI: 0000000000000003
[1262804.268423] RBP: 00007ffd83c46700 R08: 000061ed517e2010 R09: 0000000000000007
[1262804.268428] R10: 000061ed517f6ec0 R11: 0000000000000246 R12: 0000000000000005
[1262804.268433] R13: 00007ffd83c46950 R14: 00007ffd83c46770 R15: 000061ed517f6ecc
[1262804.268442]  </TASK>
[1262804.268852] OOM killer enabled.
[1262804.268856] Restarting tasks ... done.
[1262804.276219] random: crng reseeded on system resumption
[1262804.276419] PM: suspend exit
```
</details>

danielzgtg, 2026-01-07:
Here's another failure:

<details>

```text
[1495590.256834] input: IPTS 045E:099F as /devices/pci0000:00/0000:00:16.4/0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04/0000:045E:099F.0035/input/input536
[1495590.258308] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Failed to process buffer: -16
[1495590.268306] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Failed to process buffer: -16
[1495590.278323] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Failed to process buffer: -16
[1495590.314335] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Failed to process buffer: -16
[1495590.400943] ipts 0000:00:16.4-3e8d0870-271a-4208-8eb5-9acb9402ae04: Failed to process buffer: -16
[1495590.400948] Oops: general protection fault, probably for non-canonical address 0xdead000000000108: 0000 [#1] SMP NOPTI
[1495590.400953] CPU: 1 UID: 0 PID: 156898 Comm: iptsd Not tainted 6.17.1-surface-2 #2 PREEMPT(voluntary) 
[1495590.400957] Hardware name: Microsoft Corporation Surface Pro 7/Surface Pro 7, BIOS 24.109.140 07/21/2025
[1495590.400959] RIP: 0010:ipts_mei_search+0x7b/0x150 [ipts]
[1495590.400964] Code: c4 0f 84 ce 00 00 00 45 39 6c 24 10 75 ec 4c 89 f7 e8 49 08 a8 c2 4c 89 f7 e8 f1 5e ad c3 49 8b 14 24 49 8b 44 24 08 4c 89 f7 <48> 89 42 08 48 89 10 48 b8 00 01 00 00 00 00 ad de 49 89 04 24 48
[1495590.400967] RSP: 0018:ffffcdd8625cfa60 EFLAGS: 00010246
[1495590.400971] RAX: dead000000000122 RBX: ffffcdd8625cfb38 RCX: 0000000000000000
[1495590.400974] RDX: dead000000000100 RSI: 0000000000000000 RDI: ffff891c82291060
[1495590.400976] RBP: ffffcdd8625cfa88 R08: 0000000000000000 R09: 0000000000000000
[1495590.400978] R10: 0000000000000000 R11: 0000000000000000 R12: ffff891c84969c28
[1495590.400980] R13: 0000000080000006 R14: ffff891c82291060 R15: ffff891c82291030
[1495590.400982] FS:  00007ed146543780(0000) GS:ffff892054260000(0000) knlGS:0000000000000000
[1495590.400985] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[1495590.400987] CR2: 000000002721b898 CR3: 0000000112fec001 CR4: 0000000000772ef0
[1495590.400989] PKRU: 55555554
[1495590.400991] Call Trace:
[1495590.400994]  <TASK>
[1495590.400998]  ipts_mei_recv+0x124/0x160 [ipts]
[1495590.401003]  ? __pfx_autoremove_wake_function+0x10/0x10
[1495590.401009]  ipts_cmd_recv_timeout+0x34/0xc0 [ipts]
[1495590.401012]  ipts_control_send_feedback+0x89/0x110 [ipts]
[1495590.401016]  ipts_control_hid2me_feedback+0x96/0xe0 [ipts]
[1495590.401020]  ipts_eds2_raw_request+0x106/0x220 [ipts]
[1495590.401023]  ipts_hid_raw_request+0x42/0xa0 [ipts]
[1495590.401027]  __hid_hw_raw_request+0x8b/0xc0 [hid]
[1495590.401036]  hidraw_get_report+0x112/0x190 [hid]
[1495590.401042]  hidraw_ioctl+0x2ea/0x3f0 [hid]
[1495590.401048]  __x64_sys_ioctl+0xa1/0x100
[1495590.401052]  ? do_anonymous_page+0x10c/0x890
[1495590.401057]  x64_sys_call+0x13f8/0x2740
[1495590.401060]  do_syscall_64+0x7f/0x1080
[1495590.401064]  ? __handle_mm_fault+0xaed/0xfe0
[1495590.401067]  ? hidraw_ioctl+0xbf/0x3f0 [hid]
[1495590.401074]  ? count_memcg_events+0xd9/0x190
[1495590.401079]  ? handle_mm_fault+0x210/0x2f0
[1495590.401082]  ? do_user_addr_fault+0x22d/0x720
[1495590.401086]  ? x64_sys_call+0x13f8/0x2740
[1495590.401089]  ? irqentry_exit_to_user_mode+0x2e/0x290
[1495590.401093]  ? irqentry_exit+0x43/0x50
[1495590.401096]  ? clear_bhb_loop+0x30/0x80
[1495590.401098]  ? clear_bhb_loop+0x30/0x80
[1495590.401101]  entry_SYSCALL_64_after_hwframe+0x76/0x7e
[1495590.401103] RIP: 0033:0x7ed145f3287d
[1495590.401107] Code: 04 25 28 00 00 00 48 89 45 c8 31 c0 48 8d 45 10 c7 45 b0 10 00 00 00 48 89 45 b8 48 8d 45 d0 48 89 45 c0 b8 10 00 00 00 0f 05 <89> c2 3d 00 f0 ff ff 77 1a 48 8b 45 c8 64 48 2b 04 25 28 00 00 00
[1495590.401109] RSP: 002b:00007fff68d8a4d0 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[1495590.401112] RAX: ffffffffffffffda RBX: 00000000c0784807 RCX: 00007ed145f3287d
[1495590.401114] RDX: 00000000093b8e00 RSI: 00000000c0784807 RDI: 0000000000000003
[1495590.401116] RBP: 00007fff68d8a520 R08: fd5ddb083cecaab3 R09: 0000000000000030
[1495590.401118] R10: 0000000000000031 R11: 0000000000000246 R12: 00007fff68d8abf0
[1495590.401120] R13: 0000000000000078 R14: 00000000093b8e00 R15: 0000000000000106
[1495590.401124]  </TASK>
[1495590.401125] Modules linked in: tls snd_seq_dummy snd_hrtimer xt_CHECKSUM xt_MASQUERADE nft_chain_nat nf_nat r8153_ecm cdc_ether usbnet bridge stp llc r8152 mii rfcomm cmac algif_hash algif_skcipher af_alg cmdlinepart bnep spi_nor mei_hdcp ipts mei_pxp mtd snd_hda_codec_alc269 snd_hda_scodec_component snd_hda_codec_realtek_lib snd_hda_codec_generic snd_hda_intel intel_rapl_msr snd_sof_pci_intel_icl snd_sof_pci_intel_cnl snd_sof_intel_hda_generic soundwire_intel snd_sof_intel_hda_sdw_bpt snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_intel_hda_mlink snd_sof_intel_hda snd_hda_codec_hdmi soundwire_cadence snd_sof_pci snd_sof_xtensa_dsp snd_sof snd_sof_utils snd_soc_acpi_intel_match surface_gpe snd_soc_acpi_intel_sdca_quirks soundwire_generic_allocation snd_soc_acpi soundwire_bus snd_soc_sdca binfmt_misc crc8 snd_soc_avs ip6t_REJECT nf_reject_ipv6 ip6t_rt snd_soc_hda_codec ipt_REJECT nf_reject_ipv4 snd_hda_ext_core intel_uncore_frequency xt_LOG nf_log_syslog intel_uncore_frequency_common snd_hda_codec xt_multiport
[1495590.401183]  nft_limit snd_hda_core intel_tcc_cooling snd_intel_dspcfg snd_intel_sdw_acpi xt_hl x86_pkg_temp_thermal xt_limit intel_powerclamp xt_addrtype snd_hwdep xt_tcpudp coretemp xt_conntrack nf_conntrack nf_defrag_ipv6 nf_de [... cut ...]
[1495668.358720] audit: type=1400 audit(1767829356.581:1137): apparmor="DENIED" operation="file_mmap" class="file" info="Failed name lookup - disconnected path" error=-13 profile="unix-chkpwd" name="" pid=157462 comm="unix_chkpwd" requested_mask="r" denied_mask="r" fsuid=1000 ouid=1000
[1495668.914013] audit: type=1400 audit(1767829357.136:1138): apparmor="DENIED" operation="file_mmap" class="file" info="Failed name lookup - disconnected path" error=-13 profile="unix-chkpwd" name="" pid=157462 comm="unix_chkpwd" requested_mask="r" denied_mask="r" fsuid=1000 ouid=1000
```
</details>

antoniovazquezblanco, 2026-11-07:
As per https://github.com/linux-surface/kernel/pull/178#issuecomment-6447636724, it seems that the correct place to send PRs and bug reports is now https://github.com/linux-surface/kernel.

This repo should be archived or updated by the maintainers!